NicoHood/Nintendo #2: Progress question?

trevor403, 2015-01-12:
How would one go about writing the Gamecube Host program? I'd be happy to do it and merge it in if you can give me a little bit of guidance. Do you have an idea of the "dialog" between the GC and Controller to make the host? This guys, https://github.com/mbrown1413/Puyo-AI/blob/master/arduino/gamecube_control/gamecube_control.ino made a host file but he uses digital IO i like that you use an analog pin. How do i get started on this? Thanks

NicoHood, 2015-01-12:
the analoge pin is a digital pin. and i use it for digital only.

I worked on the host some month ago but it failed. the docs in the readme should guide you to anything you need. the best thing would be to hardcode the reading/sending like brownan did. the link is also in the readme. so you'd have to learn assembler to do this.

The code looks interesting but i currently have no time to update anything. I am currently working on the usb stuff. So see the readme links, there is everything you need. sorry. if you have any progress, let me know :)

trevor403, 2015-01-13:
I tried something by adapting that code but it didn't work too well. I'm using a 5v 16MHz controller with logic converter and regulator which is annoying and an extra step where error can occur. I'll come back to this when my 8MHz 3.3v controller comes in the mail. But I don't have a logic analyzer so I can't really know if I'm getting it right ahhah I'll just keep trying thanks.

NicoHood, 2015-01-13:
with a 8mhz mcu it is very difficult or maybe impossible. And without a logic analyzer its even harder. Good Luck :)

trevor403, 2015-01-13:
Anyway to run an 8MHz crystal on 3.3v? But I am a software engineer and I
have not stooped down to the level of ASM so this task may not be for me...

closed 2015-02-04 by trevor403