NicoHood/Nintendo #3: Gamepad1 and HID do not begin

trevor403, 2015-02-04:
There is an issue with compiling the Gamecube_USB_HID project. The compiling error is as follows:

```
Gamecube_USB_HID.ino: In function 'void setup()':
Gamecube_USB_HID:18: error: 'HID' was not declared in this scope
Gamecube_USB_HID:20: error: 'Gamepad1' was not declared in this scope
Gamecube_USB_HID.ino: In function 'void sendGamecubeReport()':
Gamecube_USB_HID:65: error: 'Gamepad1' was not declared in this scope
```

I checked the code under the HID project and this one and there is no initialization of Gamepad1. Finding the initialization of HID was harder but i'm not sure if it is. Any idea how to make this project work?

NicoHood, 2015-02-04:
You have to check the Gamepad examples in the HID project and update this sketch.

The funny thing is that today I looked into this Gamecube project again and felt like updating it soon again. I looked into brownans inline assembler which looks pretty good. I will try to make this work with my library as well and make it work with IDE 1.5.8.

Also because HID Project requires IDE 1.5.8 but the code is based on 1.0.5 compiler you wont be able to run this code. Or you have to use an older version of the HID Project (i guess it was version 2 or 1.8).

Things are getting more awesome soon, I hope. Feel free to help me with the assembler or simply testing the stuff. The best way you can reach me right now is the HID Project chat on github. Click the link on the Github readme of the HID Project. The message will be read even if I am currently offline. In about 2 hours I will be home, maybe we'll see each other there.

Thx for this note anyways. :)

trevor403, 2015-02-04:
Adaption of this code shouldn't be hard, there is so much work going into the HID project, just transferring the gc_send and get_gc functions into an HID shell project should do nicely? But the Nintendo project is a little more complicated, to port up to 1.5.8 Anyways I will do some work on my own and ill commit anything I can make work. Thank you for making yourself so available! Feel free to close this issue

NicoHood, 2015-02-06:
You wont be able to use the current Nintendo Project, but Brownans inline assembler should work for now. Let me know of any progress. I will close the issue when HID and Gamecube both work together again.

NicoHood, 2015-02-08:
Can you please try again with the current beta version 1.1?
https://github.com/NicoHood/Nintendo/tree/beta

Should work with the newer IDE 1.6 as well.

Will update the HID sketch asap.

NicoHood, 2015-02-10:
HID sketch is updated and tested with HoodLoader2. Should work with Leonardo as well.
Make sure to select the Gamepad core and if Windows doesnt recognize the controller, see the troubleshoot in the HID Project.

Coul you please give me a little feedback about the update? Does it work?

closed 2015-02-14 by NicoHood