NicoHood/Nintendo #4: Add 8mhz and 20mhz support

NicoHood, 2015-04-18:
For 8mhz repeat the reading loop 3 times to speed thing up (dont use the jump). For 20mhz just add more nops.

trevor403, 2016-02-03:
I know that discussion for using an 8mhz processor wasn't going to happen without timing tricks so are you saying that we want to run the read loop multiple times to make sure we don't miss anything between controller cycles? Or is this a solution to a different aspect of the issue.

NicoHood, 2016-02-03:
I can imagine of 3 solutions:

1. Use direct (hardcoded) port access. This can be accieved with some trickey templates, so the user can still use normal arduino pins. FastLED does this for example. I will use this method later on anyways, since I personally want to switch to a makefile build environment, if possible and get deeper into the code, to be more efficient and smaller in code size. But it would be possible even for Arduino and both can be used at the same time.

2. That is the solution i suggested above. At the moment we read, check the state and jump back up if its not the signal we want. (i dont remember the code well enough, but you should see it, its commented very clear). And the trick was to not use this loop and rather unroll it. Just hardcode this loop 3 times (because normally it will not run more than 3 times, within this timings its impossible.). This way we can save the jump back up and measure faster and get the timings still right for 8mhz.

That was the idea, it should be possible, but i never looked into that. I also did not test the n64 code any further, someone should really do this. Hmm maybe me as well.

Feel free to post suggestions.

NicoHood, 2016-06-19:
In am going to close this issue, since 16mhz is far enough and nobody uses 20mhz.
8 MHz would require some special tricks, mentioned above, which I do not plan to implement. Even Attinys can run at 16mhz with internal PLL so I see no reason to get 8MHz running. A hardcoded per pin implementation would suit better then. And since the code is well documented everyone can implement this if desired.

closed 2016-06-19 by NicoHood