NicoHood/Nintendo #10: Compilation problem

Dreamer68, 2017-01-26:
Hello, I have a problem when I try to compile the program ‘Gamecube_USB_HID.ino’:

"
...\Arduino\libraries\Nintendo-master\src\Gamecube_N64.c: In function 'gc_n64_send_get':

...\Arduino\libraries\Nintendo-master\src\Gamecube_N64.c:195:5: error: can't find a register in class 'POINTER_REGS' while reloading 'asm'

     asm volatile (

     ^

..\Arduino\libraries\Nintendo-master\src\Gamecube_N64.c:195:5: error: 'asm' operand has impossible constraints

lto-wrapper: ..\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-gcc returned 1 exit status

../appdata/local/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
"

Can you help me?

NicoHood, 2017-01-26:
You are using an old version of the library. How did you download it? Please update. Also update your IDE.

Dreamer68, 2017-01-26:
Yet I downloaded it yesterday here:
Https://github.com/NicoHood/Nintendo

NicoHood, 2017-01-26:
Please use the latest IDE and tell me the sketch you try to compile.

Dreamer68, 2017-01-26:
I tried with the latest version, but there is the same error.

I try to compile, ‘Gamecube_USB_HID.ino’

NicoHood, 2017-01-27:
Fixed in the latest version. Thanks for your report!
Please close this issue if the new version works for you.

Dreamer68, 2017-01-27:
Thanks.
I managed to compile it. thank you very much :)

closed 2017-01-27 by Dreamer68