NicoHood/Nintendo #11: Mayflash adapter

rosshadden, 2017-04-01:
I have this library working great with my Wii, but it does not work with my Mayflash Gamecube adapter, set to Wii U mode.  Do you have any ideas on how we can get this library to support these adapters?

NicoHood, 2017-04-01:
What are you trying to do? You can combine this lib with the HID Project and emulate your own controller. No need for a mayflash adapter.

rosshadden, 2017-04-01:
I'm trying to make something that works in all situations a normal GameCube controller does. Which hopefully includes the mayflash adapter, as that is the way the internet recommends using controllers with dolphin, and thus many people use it.

I was hoping you knew some possible starting points to try messing with in the code (device status IDs, byte sequences for `gc_init` or `gc_read`, etc.), if not the entire solution. Otherwise I might end up purchasing a logic analyzer and trying to figure out what's different between mayflash and official adapters.

Thanks!

NicoHood, 2017-04-01:
You'd need a logic analyzer. Maybe he probes for something we dont respond to yet. The gamecube itself is not that picky.

However mayflash itself is only an usb hid device. You can make your arduino and HID device directly, that be way more straight forward. I see no reason why not to do it.

rosshadden, 2017-04-02:
The Mayflash adapter has two modes toggled by a physical switch: "Wii U" and "PC".  The Wii U mode has direct support in Dolphin via the driver they ship with.  Using it in PC mode which is an HID is frowned upon in the community and does not work nearly as well as the native input.

MH-Sam, 2017-06-13:
What behaviour have you been observing if you plug an arduino modded controller (using this library) into the Mayflash 4port adapter in Wii U mode? 
My modded controller appears to be working in Dolphin with the adapter in Wii U mode.

LeagueOfDoges, 2017-07-07:
I am relatively new to all of this but hopefully my two cents might help.

I was also having trouble getting this to work with the mayflash adapter. I think that this issue is a result of how mayflash interprets gamecube controller protocol, and is probably not fixable by modifying this library. 

I think the only only fix for mayflash adapters might involve modifying or rewriting the drivers.

The easiest workaround I found was to use an official Nintendo adapter or console  instead of mayflash. It's not a good solution because it involves spending money, but it is easy. 

I'll post some more comments find a different solution.  But in the meantime I hope helps.

NicoHood, 2017-09-16:
I just debugged this issue with @SimpleControllers with the following results:

His gamecube controller (jap, emerald) answers to the init signal with a trailing 0x20 (instead of 0x30) and a stop bit of 2 us. We changed the code accordingly, so that is sends the exact same answer as the controller (debugged with a logic analyzer), but it still does not work, the console does not try to further read the controller and still queries for 0x00 (init).

The only difference is that the arduino answers within 62us and the controller within 2us. I personally only tested this on a PAL (german) gamecube (white). It is possible that the wiiu/mayflash adapter requires a way faster response than the original gamecube does.

In order to solve this issue the code would require a complete rewrite in assembly (even more than we have now) or it would require a faster MCU, but still the code needs to be ported.

NicoHood, 2017-09-21:
A user reported to get the mayflash adapter working after a firmware upgrade. He has the following device:
http://www.mayflash.com/products/nintendowiiu/w012.html

Here is the direct download link to the firmware V05 that he used (warning: insecure http used):
www.mayflash.com/File.asp?Id=106

AllStarWeeb, 2017-10-19:
I figured it out. I had originally tested using the sample from here that closed with `if(!console.write(data)) delay(100);`
but that condition never returns true so it would never write. Simply using console.write(data); instantly fixed it and that is the only change I had to do.
Updating the firmware has nothing to do with it.

SimpleControllers, 2017-10-19:
That's not it.  That is part of the solution though.  I tried removing that before updating the adapter (just having the pure write statement) but no luck .  It was only until after I updated the adapter and removed the ifnot statement till it started working.

AllStarWeeb, 2017-10-19:
But I didn't update the adapter. 😕

SimpleControllers, 2017-10-19:
When did you buy your adapter?

AllStarWeeb, 2017-10-19:
If I remember right, near this time last year.

SimpleControllers, 2017-10-19:
Interesting, I did all of my testing without the if not statement and tried to get mayflash to work countless times and it would not work regardless of what I tried (as well as some other users, they got other thirdparty adapters to work by removing the ifnot but never mayflash).  Maybe mine was a super old version from an old batch or something no idea?  I got mine around the same time as you, so that is really odd.

AllStarWeeb, 2017-10-19:
Weird... Yeah I spent a couple hours the last few days debugging this because I'm modding the DK Bongos. Thought I'd try my luck here. By the way, nice tutorial. Helped a lot :)

luchris429, 2020-01-25:
> A user reported to get the mayflash adapter working after a firmware upgrade. He has the following device:
> http://www.mayflash.com/products/nintendowiiu/w012.html
> 
> Here is the direct download link to the firmware V05 that he used (warning: insecure http used):
> [www.mayflash.com/File.asp?Id=106](http://www.mayflash.com/File.asp?Id=106)
> 
> Can anyone verify this please?

I tried this and it did not work. However, I'm also doing this on MacOS which may add some complications (I am using the codeless .kext here https://wiki.dolphin-emu.org/index.php?title=How_to_use_the_Official_GameCube_Controller_Adapter_for_Wii_U_in_Dolphin#OS_X).

NicoHood, 2021-03-06:
What is the current status of this in 2021? Can anyone give me a short update if the issue can be resolved with the firmware update or do we need to patch something in this arduino library?

WinstonPHz, 2021-03-14:
I can confirm that it works with Version 5.0 of the firmware.

WinstonPHz, 2025-09-28:
Can someone check the Version 9.0 of the firmware?  I am not able to get the mayflash to read the data from my arduino with this version.
I was changing a lot of things so it might be one of those things but it has not worked sense updating the firmware.

WinstonPHz, 2025-09-30:
I was able to check and Version 9.0 Does not work. Stay on 5.0