NicoHood/Nintendo #14: Wrong Input on gamecube controller whuke using mayflash adapter on Wii U

Koloss92, 2018-01-08:
Hi,

First, thanks for your work, it's really appreciated!
Second, I've made my own custom gamecube controller. 
-It work great in my PC (gamecube to usb) when I play 
All my gamecube game.
-It work great too on Wii w/ gamecube port AND my
Original gamecube, every input working like a normal controller 

.. Where I have a problem is when I use a mayflash adapter 
(or another one, thats not working with everyone else) on the Wii u
My input are not the same, I can't use my trigger (button L and R)
They're not working and I have only X or Y button working, the
Other input work as a Z button (which make 2 Z instead of 1)

Can you clarify how I can make it work on it? Thanks!!

NicoHood, 2018-01-09:
That is not very precise, I have no idea.

Koloss92, 2018-01-10:
[image]

This image say everything about my problem ! It's more precise ?

NicoHood, 2018-01-10:
So you want to emulate a gamecube controller to the wiiu with a mayflash adapter? This could help:
https://github.com/NicoHood/Nintendo/issues/11#issuecomment-331177372

Koloss92, 2018-01-12:
Right, but the problem persist, the mayflash adapter have not problem when  it's connected to the PC, it's only when I'm playing on the Wii U where the problem start, I'm not sure why the input change, however, I'm stall just because of that... :/

NicoHood, 2018-01-12:
You'd need a logic analyzer then to debug the problem.

If you buy me one of those adapters, I might be able to test. But I cannot promise whenever I will do that. But I got a wiiu and a nintendo switch. It will take me some time to set it all up again, find the issue and especially find free time.

Koloss92, 2018-01-26:
Sorry for the late answer, what I want is simply ; I just need the same input as the Wii for the Wii U, except if I need a switch, you can correct me if im wrong

NicoHood, 2018-01-27:
I have absolutely no idea on your problem. Use a logic analyzer and find out the root cause.

closed 2018-02-08 by Koloss92