NicoHood/Nintendo #15: [Request] Support for Teensy Boards

TheRealDxRed, 2018-01-26:
As the title says, I would like to know if it's possible for this library to be ported to Teensy boards. Specifically the Teensy LC.

LeagueOfDoges, 2018-01-26:
I have been working on support for teensy boards

LeagueOfDoges, 2018-01-26:
the whole backend of this library is written in avr assembly, porting it to teensy would require at least all of the timing to be rewritten. 

I'm not much of a fan of assembly, and I wanted something a bit more powerful than arduino, so I started writing my own library for teensy. It is almost functional... let me know if you are interested

TheRealDxRed, 2018-01-26:
Thanks for the quick response! I would love to see what you've made.

NicoHood, 2018-01-27:
@LeagueOfDoges is right, it requires lots of changes. If you got something ready, please let me know, so I can link to it in the readme. I will not do that implementation myself.

Machine-Dynamics, 2018-06-26:
I know it's been a while, but I just stumbled onto this, and having this library work with Teensy would save me a lot of headache. Did you ever get this library working, @LeagueOfDoges ?

DogeSSBM, 2018-06-26:
@Machine-Dynamics https://github.com/DogeSSBM/DogeBawx

HiramSilvey, 2021-06-30:
I know this is an old thread, but what's the status of the suggested replacement library @DogeSSBM? I'm specifically interested in running this on a Teensy 4.0 or 4.1. Thanks!

DogeSSBM, 2021-06-30:
I got one working for my purposes on the Teensy 3.2/3.6. I'm am unsure how it would work on the 4.0/4.1, but I'd be willing to help you get it ported. 

https://github.com/DogeSSBM/Dogebawx