NicoHood/Nintendo #16: Gamecube controller to Wii U

Koloss92, 2018-02-08:
Hi !

Do you a code to make work gamecube controller on Wii U ?

Thanks !

DogeSSBM, 2018-02-08:
@Koloss92 I'm not exactly sure what you are asking. But this library works perfectly for me on the Nintendo Gamecube Controller adapter for Wii U, and the Mayflash 4 Port Gamecube Controller adapter for Wii U.

Koloss92, 2018-02-08:
I've made a customed controller, probably my coding have some issues, but I can't use it on my Wii U, some input are wrong !

DogeSSBM, 2018-02-08:
@Koloss92 you already opened an issue on that topic. This library is 100% accurate as far as the protocol goes. My guess is that your trying to use the digital portion of the L and R triggers. Try using the analog ones instead. 

In Super Smash Bros. for Wii U, the analog portion of the trigger is used to shield, not the digital press. As for the issues with the Z button, google "debounce".

 Include your source code and wiring diagrams / pictures if you want me to help any further.

Koloss92, 2018-02-08:
Yeah the last one was a little bit hard to understand, that's why I've open a new one there.

Here's the code and the board

[SmashBox_01_07.zip](https://github.com/NicoHood/Nintendo/files/1708797/SmashBox_01_07.zip)
[image]

Koloss92, 2018-02-11:
@DogeSSBM Still there for some help ? :)

NicoHood, 2018-04-15:
I will close this issue, as the library works properly.

closed 2018-04-15 by NicoHood