NicoHood/Nintendo #18: GC rumble question

clbr, 2018-08-23:
Your GC page mentions the rumble motor drew 30 mA. This is quite low, and I can't find any other sources online with measurements. Other consoles' controllers seem to draw more.

So just asking to be sure, is it 30 mA and not 300 mA for example?

NicoHood, 2018-08-25:
I have no idea. I think i measured it with my multimeter, but that could be inaccurate. You might want to find it out yourself.

clbr, 2019-01-18:
I got to measuring three GC controllers' 5V lines when rumbling. Results:
Original Nintendo: 36 mA
Smash Ultimate (released 2018 Dec): 32 mA
3rd party, GameStop brand: 40 mA

So they indeed draw so little.

NicoHood, 2021-01-22:
I've added this issue to the docs as a link.

closed 2021-01-22 by NicoHood