NicoHood/Nintendo #21: Write Multiple GC Consoles

3lacksky, 2019-05-02:


3lacksky, 2019-05-02:
First, sorry for not adding a description at the top.
I have a problem reading 2 Controllers and write this data to 2 ports of the Console.
If i read 2 Controllers and just write one of the two, it works, but writing 2 just does nothing, hope you can help.

#include "Nintendo.h"
CGamecubeController GamecubeController2(4);
CGamecubeController GamecubeController1(7);
CGamecubeConsole GamecubeConsole1(6);
CGamecubeConsole GamecubeConsole2(5);
void setup() {
}

void loop() {
GamecubeController1.read();
GamecubeController2.read();
GamecubeConsole1.write(GamecubeController1);
GamecubeConsole2.write(GamecubeController2);
}

NicoHood, 2019-05-02:
It is possibly a timing issue and cannot be solved with an arduino. Its too slow.

3lacksky, 2019-05-02:
Ok, thank you, I will try out more and maybe it will work somehow :)

closed 2019-05-02 by 3lacksky

NicoHood, 2019-05-04:
Let me know, if you have any solution. good luck :-)