NicoHood/Nintendo #22: N64 need a delay between begin() and read()

Danjovic, 2019-07-17:
I was always getting and error while running N64Controller.ino example. 
While troubleshooting the code I have found that if you went too fast between sending the initialization command (0x00) on begin() function to sending a request 0x01 on read() function the controller does not answer to the request.
The standard example code takes 23us from end of answer to command 0x00 to begin of command 0x01.
I have modified the example (see code below) and experimented some delay values and checked the output using a saleae logic analyzer.
```
void loop()
{
    N64Controller.begin();
    delayMicroseconds(100);  // Added delay
  // Try to read the controller data
  if (N64Controller.read())
  {
    // Print Controller information
```
with the added delay I still had issues while the total time was under 80us.

Then I suggest you to add such a delay right before returning on  CN64Controller::begin(void) function or at least mention that issue in documentation.

NicoHood, 2019-07-30:
Thanks for that suggestion!
I try to fix it soon.

NicoHood, 2019-09-08:
Sorry for the late reply.

I've checked the code, but it does not contain a begin() function at all. So Why did you add it? Is it even required? Or do you just need some delay inside the loop? What if you move the begin() to the setup) function?

Danjovic, 2019-09-14:
Hi
The begin() method is inside the CN64Controller class (in N64API.h) . Such method is called implicitly by the read() method when the property 'Connected' is not true.
My N64 controller refuses to answer without a short delay between the initialization (gc_init() ) and the first read (n64_read() ). I think the controller need that time to properly initialize itself.
The loop() worked fine without much delay between consecutive readings.
To make the library built-in example work I just added a call to .begin() method followed by a delayMicroseconds(100) on the arduino setup()

NicoHood, 2021-01-22:
@JonnyHaystack Can you maybe help here? I guess you are currently more into the code than me...

JonnyHaystack, 2021-01-23:
Not sure why this would be a problem sorry. I don't have a setup for testing with an N64 controller either

NicoHood, 2021-03-06:
Thank you guys! I've fixed the issue now. Please check that out.

commit 4e63a18d (2021-03-06): Fix #22 add delay between begin() and read() for N64 controller
closed 2021-03-06 by NicoHood