NicoHood/Nintendo #26: It sometimes does not send

kahveciderin, 2020-09-05:
Sometimes i need to hold a button for it to send the data to the console. Sometimes it doesn't work at all.

My code:
`
#include "Nintendo.h"

CGamecubeConsole GamecubeConsole1(8);      //Defines a "Gamecube Console" sending data to the console on pin 8
Gamecube_Data_t d = defaultGamecubeData;   //Structure for data to be sent to console

//This is needed but you don't need a controller on pin 7
//CGamecubeController GamecubeController1(7);
// Pin definitions
#define pinLed LED_BUILTIN

void setup()
{
  // Set up debug led
  pinMode(pinLed, OUTPUT);

  // Start debug serial
  Serial.begin(115200);
  //GamecubeController1.read();

    d.report.a = 0;
    d.report.b = 0;
    d.report.x = 0;
    d.report.y = 0;
    d.report.z = 0;
    d.report.start = 0;
    d.report.r = 0;
    d.report.l = 0;
    d.report.left = 0;
    d.report.xAxis = 128;
    d.report.yAxis = 128;
    d.report.cxAxis = 128;
    d.report.cyAxis = 128;

    digitalWrite(pinLed,LOW);
    for(int i = 0; i < 15; i++){
      GamecubeConsole1.write(d);
    }
    digitalWrite(pinLed,HIGH);

  pinMode(13, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(11, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
}

void loop()
{

  
    
    //reports data
    d.report.a = !digitalRead(10);
    d.report.b = !digitalRead(11);
    d.report.x = !digitalRead(12);
    d.report.y = !digitalRead(13);
    d.report.z = 0;
    d.report.start = 0;
    d.report.r = 0;
    d.report.l = 0;
    d.report.left = 0;
    d.report.xAxis = 128;
    d.report.yAxis = 128;
    d.report.cxAxis = 128;
    d.report.cyAxis = 128;
   
    GamecubeConsole1.write(d);
    
    Serial.print("Data sent: ");
    Serial.print(d.report.a);
    Serial.print(d.report.b);
    Serial.print(d.report.x);
    Serial.println(d.report.y);
    delay(1);
}`

NicoHood, 2020-10-01:
That is because your loop is too slow. Do not use the serial, and the digitalread might also be too slow - use direct port manipulation.

1. The console asks for the button states
2. You do not reply, as you are sending data via serial
3. The console thinks the controller disconnected
4. The console will retry to connect
5. You might be sending via serial again
6. Eventually the console will be able to read a correct button state
7. But you are sending via serial again and the console thinks the controller disconnected again,

You can test this very good with mario football, which will pause the game if the controller disconnects.

kahveciderin, 2020-10-24:
It still does not send @NicoHood

NicoHood, 2020-10-28:
What did you change?

kahveciderin, 2020-10-30:
I removed the serial and digitalreads

NicoHood, 2020-10-31:
Can you post the sketch again?

NicoHood, 2021-03-06:
Ping

sirjeannot, 2021-03-11:
Hello. I had the same issue on my bluetooth ps3 controller for gamecube projet.
Calling GamecubeConsole1.write(d); on every loop cause the error writing to gamecube. Only call it if you detect a change in the input.

closed 2021-03-12 by NicoHood

kahveciderin, 2021-08-25:
update: I solved this issue by only sending the data when a key has changed state