NicoHood/Nintendo #39: No input MKWii

SpicyCheesePizza, 2021-06-17:
I currently use a B0XX-style controller running this backend, and as the title of this post suggests I don't get any inputs read in MarioKart Wii as a gamecube controller when I'm plugged into the Wii. There aren't any other issues with the cable or other games (or even the same setup in dolphin), only MKWii on its native console. Is this something that can be handled by this backend or is there some game restriction that I am unaware of? 

My current guess as to what's going on is that MKWii requires some specific form of input that needs to match the gamecube controller exactly, is there any merit to this or is it more likely some other issue?

NicoHood, 2021-06-17:
That is a good question. I dont know. It can be, that the library responds too slow. Depending on your code in loop, it might be too slow. The polling is game dependent, not console dependent from my observations. 

It is very unlikely that mario kart uses a different polling format. The library already supports a lot. You'd need to check that yourself with a logic analyzer.

Does this example work?
https://github.com/NicoHood/Nintendo/blob/master/examples/Gamecube/GamecubeConsole/GamecubeConsole.ino

Skuzee, 2021-06-17:
I believe the polling to be console dependent as well. The controller just responds to the commands sent via the console. I also have seen conflicting times about update frequency; This leads me to believe that it is game dependent.

In Wii Virtual Console version of Ocarina of Time:
- The console polls the controller twice quickly with about 1.080ms between updates, then waits 15.640ms and polls twice again.
- It takes about 0.380ms for the console to send the update command, and the controller to respond with data over serial. That happens twice: once between controller and micro-controller, and again between micro-controller and Wii. My closest estimate would mean only 0.320ms left to finish any other code you have in your loop. Which is still 'plenty' of time to do any calculations, as long as you don't halt the processor. 

It's possible that your code is introducing significant delays? You can buy a $5 usb data analyzer that would give you more insight to the timing. I have a 24MHZ 8 channel one that works great for the price.

NicoHood, 2021-06-17:
I also remember that the console is reading the data twice in a very short time. I thought that I've implemented that in my code, but I cannot find any of this. But maybe I am just not searching good enough.

So the idea to solve this issue is to try send the data twice:
```
GamecubeConsole1.write(GamecubeController1);
GamecubeConsole1.write(GamecubeController1);
```

NicoHood, 2021-07-09:
@SpicyCheesePizza did you manage to solve your issue?