NicoHood/Nintendo #40: Help with the setup using a different Logic Level Converter

vincentadamthefirst, 2021-06-24:


Skuzee, 2021-06-24:
Looking at the data sheet I think you might need to do something with the OE output enable pin? I think it's active high, so it should be pulled high to enable. You could use another arduino pin, or maybe just hardwire it. 

The manual also says:

> "When the output-enable(OE) input is low, all outputs are placed in the high-impedance(Hi-Z)state.
> To ensure the Hi-Z state during power-up or power-down periods, tie OE to GND through a pull-down resistor.The minimum value of  the resistor is determined by the current-sourcing capability of the driver."

I don't think it's required to pull-down OE on startup, but I assume it's to avoid transmitting garbage. So it would be safety to pull it down with a resistor and then pull it high with an arduino pin. 

You mentioned that you only focused on the LLC wiring, but I'm gonna say, make sure the controller shares 3.3v and ground. Ideally you would use the 3.3v from the wii console for the VA and it would be isolated from the arduino.

vincentadamthefirst, 2021-06-25:
I tried what @Skuzee suggested by using another 1k resistor and pulling the OE to GND using a jumper cable. After powering the Arduino I plugged the jumper cable to 3.3V (also tried 5V) but it did not work.

As for the wiring, I currently only want to read the input of a GC controller. I use the 3.3V of the arduino to power VA.

closed 2021-07-08 by NicoHood