NicoHood/Nintendo #54: Help with my Pro Micro setup?

lipe-dev, 2022-08-10:
Hey guys and girls. Can someone help me troubleshoot this setup

I'm trying to use this Pro Micro because it is what I had in hand. I tried to replicate the Uno example as closely as I could.

My Pro Micro does not have a 3v3 power pin, so I also added a 3.3 voltage regulator.
Everything else mimics the example Fritzing file as closely as I could.

The only real difference here than the reference images is that on my extension cable the blue and brown cables are switched around, as you can see, blue goes to pin 4 and brown goes to 6.

I remember @Skuzee mentioning being able to make this work on a pro micro on my other issue, but I'm currently unable to.

When running the example programs, all it says is "Error reading Gamecube controller." over and over :(

My guess is that the pin 4 I'm using doesn't work for this for some reason? And what is the switch from pin 8 to ground for?

[image]
?

Skuzee, 2022-08-10:
At first glance your setup looks good. Here are some things that I would check:  

- Consider connecting the 5v and 3.3v GNDs together. Not sure what type of regulator that is, but if it's a floating ground then that might be messing up the data line. (Since the data it's referenced off the 5v ground)
- Make sure the logic level converters you bought are bidirectional. 
- Make sure your pinout to your controller is correct. double check those colors!  
- It might be a software issue? Make sure the pin number is set correctly.

lipe-dev, 2022-08-10:
> At first glance your setup looks good. Here are some things that I would check:
>
> * Consider connecting the 5v and 3.3v GNDs together. Not sure what type of regulator that is, but if it's a floating [...]

Thanks for the help.

About the software, in the example I tried running I did not see any lines setting the pins, and I also did not find anything documented about it. I just used pin 4 because that's what the image in the docs had.

I'll try connecting the grounds together.

The logic level conevrter I got are both single and bidirectional, with the setup I have it is in bidirectional mode.

Skuzee, 2022-08-10:
Can you share/link the firmware you're using?  
You might also want to find out what that switch is for on pin 8?  
The pin is set when the controller is defined at the top of the program: `CGamecubeController GamecubeController(7); // pin number`

lipe-dev, 2022-08-10:
``> Can you share/link the firmware you're using? You might also want to find out what that switch is for on pin 8? The pin is set when the controller is defined at the top of the program: `CGamecubeController GamecubeController(7); // pin number`

I switched to this one now:

```
/*
  Copyright (c) 2014-2016 NicoHood
  See the readme for credit to other people.

  GamecubeController example
  Enter any key into the serial monitor to start reading.
  Press A on the controller to enable rumble.
*/

#include "Nintendo.h"

// Define a Gamecube Controller
CGamecubeController GamecubeController(4);

// Pin definitions
#define pinLed LED_BUILTIN

void setup()
{
  // Set up debug led
  pinMode(pinLed, OUTPUT);

  // Start debug serial
  while (!Serial);
  Serial.begin(115200);

  // Wait for some Serial input
  Serial.println(F("Enter any key to start reading the controller."));
  Serial.println(F("Press A on the controller to enable rumble."));
  while (Serial.read() == -1);

  Serial.println(F("Starting controller emulation"));
  Serial.println();
}

void loop()
{
  // Try to read the controller data
  if (GamecubeController.read())
  {
    // Print Controller information
    auto status = GamecubeController.getStatus();
    auto report = GamecubeController.getReport();
    print_gc_report(report, status);
    delay(100);

    // Rumble if button "A" was pressed
    if (report.a) {
      GamecubeController.setRumble(true);
    }
    else {
      GamecubeController.setRumble(false);
    }
  }
  else
  {
    // Add debounce if reading failed
    Serial.println(F("Error reading Gamecube controller."));
    digitalWrite(pinLed, HIGH);
    delay(1000);
  }
  digitalWrite(pinLed, LOW);
}

void print_gc_report(Gamecube_Report_t &gc_report, Gamecube_Status_t &gc_status)
{
  // Print device information
  Serial.print(F("Device: "));
  switch (gc_status.device) {
    case NINTENDO_DEVICE_GC_NONE:
      Serial.println(F("No Gamecube Controller found!"));
      break;
    case NINTENDO_DEVICE_GC_WIRED:
      Serial.println(F("Original Nintendo Gamecube Controller"));
      break;

    default:
      Serial.print(F("Unknown "));
      Serial.println(gc_status.device, HEX);
      break;
  }

  // Print rumble state
  Serial.print(F("Rumble "));
  if (gc_status.rumble)
    Serial.println(F("on"));
  else
    Serial.println(F("off"));

  // Prints the raw data from the controller
  Serial.println();
  Serial.println(F("Printing Gamecube controller report:"));
  Serial.print(F("Start:\t"));
  Serial.println(gc_report.start);

  Serial.print(F("Y:\t"));
  Serial.println(gc_report.y);

  Serial.print(F("X:\t"));
  Serial.println(gc_report.x);

  Serial.print(F("B:\t"));
  Serial.println(gc_report.b);

  Serial.print(F("A:\t"));
  Serial.println(gc_report.a);

  Serial.print(F("L:\t"));
  Serial.println(gc_report.l);
  Serial.print(F("R:\t"));
  Serial.println(gc_report.r);
  Serial.print(F("Z:\t"));
  Serial.println(gc_report.z);

  Serial.print(F("Dup:\t"));
  Serial.println(gc_report.dup);
  Serial.print(F("Ddown:\t"));
  Serial.println(gc_report.ddown);
  Serial.print(F("Dright:\t"));
  Serial.println(gc_report.dright);
  Serial.print(F("Dleft:\t"));
  Serial.println(gc_report.dleft);

  Serial.print(F("xAxis:\t"));
  Serial.println(gc_report.xAxis, DEC);
  Serial.print(F("yAxis:\t"));
  Serial.println(gc_report.yAxis, DEC);

  Serial.print(F("cxAxis:\t"));
  Serial.println(gc_report.cxAxis, DEC);
  Serial.print(F("cyAxis:\t"));
  Serial.println(gc_report.cyAxis, DEC);

  Serial.print(F("L:\t"));
  Serial.println(gc_report.left, DEC);
  Serial.print(F("R:\t"));
  Serial.println(gc_report.right, DEC);
  Serial.println();
}
```

set to pin 4

lipe-dev, 2022-08-11:
@Skuzee I just hooked it up to my other pro micro, same results :(

I'm starting to think I'll need to go to a hacker space or something to use an oscilloscope to see if signals are actually flowing around as they should.

lipe-dev, 2022-08-11:
Solved. The solution was so incredibly simple and such an incredibly dumb mistake on my end. 
I asked for help here https://www.reddit.com/r/arduino/comments/wle3h3/has_anyone_had_success_reading_a_gamecube/

The voltage rails on the sides are not connected all the way through, I can't believe I did not know this!

Thanks and sorry for wasting your time @Skuzee

closed 2022-08-11 by lipe-dev