NicoHood/Nintendo #55: Anyway this can be ported to work on the Switch??

vin2890, 2022-10-28:


Skuzee, 2022-10-28:
Short answer: not without a lot of work or extra parts. 

The communication part of the library is written in assembly for 16mhz avr boards. So you would need to use an external chip of some kind to convert the data to USB or Bluetooth or something the switch can interface with.  

Perhaps if there was a way to write custom code for a modded switch then you could do it more directly? There's a similar library that's written in c that might help you on your journey.  
https://github.com/DogeSSBM/DogeBawx

vin2890, 2022-10-28:
I think this guy has been able to communicate through ESP32 using the gamecube controller, I'm wondering if we program the arduino board with some of the same functions, if we can get at least the switch to recognize and communicate with the gamecube controller

https://github.com/NathanReeves/BlueCubeMod/blob/master/Firmware/BlueCubeModv2/main/main.c

vin2890, 2022-10-28:
Anyway we can convert the above code to arduino??

Skuzee, 2022-10-30:
Due to the limiting processing power of a 16MHz arduino and its lack of wifi/bt connectivity (not to mention the chip shortage increasing its cost) it would probably be better just to use an esp32 or some other microcontroller for the project. You would not be able to use this library exactly, but it is probably a better platform. That said I don't want to dissuade you if you do choose to use this library with an arduino.

vin2890, 2022-10-30:
I'm okay with using the ESP32, The only thing I am really trying to do with this library is to map a 2 frame input to a single button, I want to be able to push Left or right on the Dpad and instead of a taunt being executed in Smash Ultimate, I would like for the 2 Frame'd Attack to be executed...The above library has everything needed to talk to the switch, But I am lost as to where I can do the button mapping

NicoHood, 2022-11-20:
Cant you just use the switch usb to gamecube adapter??

vin2890, 2022-11-21:
> Cant you just use the switch usb to gamecube adapter??

I have tried that with this library, But I believe the Switch and gamecube communicate differently, maybe the sequence of commands are different....I have not been able to get the controller to send out any commands, when using it and this library...This guy was able to get correct communcation using bluetooth, I wonder if the commands can be used to update the library to include switch capability => https://github.com/NathanReeves/BlueCubeMod/blob/master/Firmware/BlueCubeModv2/main/main.c

NicoHood, 2022-11-21:
Bluetooth is totally out of scope of this library. It is a whole different protocol

vin2890, 2022-11-21:
I get that its out of Scope, I'm just saying the above library is sending a set of commands over bluetooth to the switch, I am wondering if we could strip the communication that the bluetooth chip uses to communicate with the switch and incorporate that with your library somehow...Its probably not much but the switch is polling the controller somehow, but not in the same way as the gamecube does, there has to be someway to do update this library to talk with the switch instead of a gamecube