NicoHood/Nintendo #70: [Gameboy Player] Unable to send controller inputs to Gamecube when playing a GBA game.

andrew171717, 2024-09-05:
Testing with the GamecubeConsole.ino example, inputs are received by games such as Smash Bros Melee perfectly, but when attempting to play the Gameboy Player with a GBA game, the controller read and console write functions start randomly failing and random inputs appear to be being received by the console. The controller will sometimes vibrate randomly as well, even though the game should not be sending a vibrate command.

andrew171717, 2024-09-06:
Writing the inputs twice back to back appears to fix this? I'll need to test more to be sure

andrew171717, 2024-09-06:
I'm having trouble making sense of this. I'll be able to send 2-4 inputs and then nothing for a few seconds and then I can send 2-4 inputs again.

Skuzee, 2024-09-06:
It's possible you're filling the serial buffer too fast and halting the processor. If the serial buffer is full it will wait until there is room to add it. You could blink a light or something to see if one of the arduinos are stopping.

This might help if you add it work, but it might drop data.

``` C++
// this example is for the Serial.print
void tryPrint(String dataPackage) { // will only write to the serial buffer if there is room. 
  if ( Serial.availableForWrite() >= dataPackage.length())
    Serial.print(dataPackage); // send data here.
}
```

andrew171717, 2024-09-06:
Bare with me, I've been at this for a while today. The serial buffer shouldn't be the problem in this case, I'm only writing controller data to the Arduino upon requesting it (which is only 8 bytes). And I don't request again until it either times out or gets the data. If it times out, I clear the buffer and throw the led high for a seconds. I have not noticed this happening.

I've tried reworking how I'm getting and sending controller inputs multiple times. I tried sending previous controller reports if I didn't get a new input in a certain amount of microseconds. But it seems like calling GamecubeConsole.write function mid serial message causes some data to be lost. 

I was starting to lose hope and randomly tried running the GamecubeConsole.write function more than 2 times (as I read in another discussion some games needed more than one write). I noticed improvments, so I tried more. This makes no sense to me, but maybe it will make sense to someone else. The more times I run GamecubeConsole.write the better my controls got. Running it about 10 times for each controller read allows me to have control over the game for about 3-5 seconds and then I lose control again for about the same amount of time. I kept trying higher and higher write amounts. Running GamecubeConsole.write 80 times seems to be close to the magic number for the Gameboy Player, at least in my scenario. Over 100 causes a little stuttering in control input, only 60 also causes some stuttering.

This monster of a loop seems to solve my problems with the Gameboy Player
```
for (uint8_t i = 0; i < 80; i++){
      GamecubeConsole.write(report);
}
```

Here is the full code for my project, I have a host sending the inputs to this Arduino over Serial for anyone who is a bit confused on the random Serial code. This doesn't work in Smash Bros Melee or the Gamecube Dashboard from my testing, but it works perfect for the Gameboy Player. I'm thinking of adding a dial to my breadboard to allow for adjusting the write amount more easily since it seems to differ between games.

```
#include "Nintendo.h"
#define WAITLIMIT 1000

// Define a Gamecube Console
CGamecubeConsole GamecubeConsole(2);
Gamecube_Report_t report;

//////////////////////////////////////////// Arduino Uno x4 ////////////////////////////////////////////
/////////////////////////////////////////  Serial to Console ///////////////////////////////////////////
void setup(){
  // Set up debug led
  pinMode(LED_BUILTIN, OUTPUT);

  // Set up serial connection to Mega
  Serial.begin(250000);
}

void loop(){
  Serial.write(true); // Request controller data
  unsigned long startTime = millis(); // Get the current time
  while (Serial.available() < sizeof(Gamecube_Report_t)){ // Wait until we get controller data
    if (millis() - startTime > WAITLIMIT) { // Timeout if we do not recieve the data
      clearBuffer(); // Clear out buffer if anything is there
      digitalWrite(LED_BUILTIN, HIGH);
      delay(1000);
      break; // Exit the loop on a failed request
    }
  }; 

  // Read the controller
  if (Serial.available() == sizeof(Gamecube_Report_t)) {
    uint8_t buffer[sizeof(Gamecube_Report_t)];
    Serial.readBytes(buffer, sizeof(buffer));
    memcpy(&report, buffer, sizeof(Gamecube_Report_t));
    for (uint8_t i = 0; i < 80; i++){
      GamecubeConsole.write(report);
    }
    digitalWrite(LED_BUILTIN, LOW);
  }
  else{
    digitalWrite(LED_BUILTIN, HIGH);
    clearBuffer(); // Corrupt data if larger than 1 report. Clear the buffer
    delay(1000);
  }
}

void clearBuffer(){
  while(Serial.available() > 0){ Serial.read(); }
}
```

andrew171717, 2024-09-07:
Also worth noting, Pokemon Box also requires around 80 writes per loop. It's not as smooth as the 80 writes to the Gameboy Player, but only writing one or two times is not enough. Furthermore, the writes to the game fail every 0.5 seconds or so, I suspect these games are polling the controller at an undocumented rate in regards to this library (this is speculation on my part). The work around I am using will have to do for now.

andrew171717, 2024-10-17:
I was finally able to get consist inputs that are not turbo by writing inputs roughly every 50 ms. To achieve this, I'm using the <TimerOne.h> class to constantly send the inputs every 800 clock cycles. It works flawlessly in the Gameboy Player, but no where else.